rcarlyle/StepperSim #34: Feature: Trinamic TMC2100

trilitheum, 2016-09-01:
I have been using the DRV8825 and the A4988 for a long time. I have a set of the TMC2100 drivers on the way and would be really interested in how their behaviour compares. 

I am fairly well acquainted with the TMC2100 datasheet, my electromechanics is pretty rusty so I don't know if I can be much help with the simulation.

rcarlyle, 2016-09-01:
Last time I checked the TMC2100 sheet, I don't think the description of the adaptive decay algorithm (SpreadCycle) was detailed enough to make a complete model. So it's hard and might require quite a bit of guesswork and oscilloscope monitoring to check the results. I don't really have time to do either at the moment. 

The main thing is figuring out the exact timing of how the on/slow/fast/slow sequence changes with coil conditions. I've considered comparing multiple TMC datasheets for drivers that use SpreadCycle to see if they have enough info together, but haven't tried it. 

In THEORY, a well-designed adaptive decay driver like the TMC2100, DRV8846, or A5984 should have pretty close to "perfect" current control. It may be temporarily wrong and thus over/under-shoot for a few PWM cycles, but should quickly self-correct and get to the desired current waveform. So, there's a workaround for not having sheets for those drivers: whenever you want to check the drivetrain performance an adaptive driver, just plug it into another sheet that has nearly "perfect" current control, like the THB6128. It should get you pretty close to the right answer. No promised though. 

The silent StealthChop mode on the TMC2100 is out of the question to simulate, it's a fundamentally different control scheme from the current PWM in the other drivers.

trilitheum, 2016-09-01:
I agree the StealthChop is a completely different control scheme, and really it isn't that useful for most applications with the positional error and the low available torque.

When my drivers arrive I will take a look with my oscilloscope and see how they compare to the DRV8825 or the THB6128.