rcarlyle/StepperSim #42: Excel viewer is unable to open

paboman, 2017-05-16:
MS excel viewer is unable to load / visualize any of downloaded .xlsx.
It seems to load in the beginning then it stops without opening any doc and waiting for user input.

open calc is also not able to load.
any advice on a free viewer than can be used ?
thx

rcarlyle, 2017-05-16:
It works for me in the last two versions of MS Excel. What version are you using? Are you downloading the raw file and then opening it, or trying to view in GitHub?

LibreOffice has been reported to work.

paboman, 2017-05-16:
the files are downloaded correctly.
I'm using the last available Ms excel viewer, which maybe cannot open the files since that version is a viewer only... i Will try with  LibreOffice